FHW-Appel/MicroMouse-shield #5: Visualisierung Fahrzeugdaten

MathisFeldmann, 2025-04-10:
**Beschreibung**
Damit wir wissen, ob das Auto z.B. angeschaltet ist, ist es notwendig eine Einrichtung zu bauen, um Fahrzeugdaten auszugeben. Dies kann z.B. durch eine LED realisiert werden. Andernfalls kann man sich überlegen, ob man nicht ein kleines Display einbaut, um sämtliche Fahrzeugdaten auszugeben. Falls man sich für das Display entscheidet, kann man auch Fehlermeldungen o.ä. auslesen.

**Akzeptanzkriterien**
- Fahrzeugzustand (eingeschaltet / ausgeschaltet)
- Akkuzustand
- Befindet es sich in einem "Run"
- (Fehlermeldungen)

MathisFeldmann, 2025-04-23:
## 🖥️ Anzeigeelement: Entscheidung für ein Display

Wir haben überlegt, ob wir zur Anzeige von Informationen **LEDs** oder ein **Display** verwenden.  
Entschieden haben wir uns für ein **Display**, da es einfacher ist, dort konkrete Informationen direkt abzulesen – ohne erst zu wissen, welche LED welche Bedeutung hat.

### 📺 Verwendetes Display

Ein kompaktes LCD mit 1,3" Bildschirmdiagonale und einer Auflösung von 240 × 240 Pixeln. Ideal für kleine, grafische Ausgaben und Statusanzeigen.

🔗 [Zum Produkt bei Reichelt](https://www.reichelt.de/de/de/shop/produkt/entwicklerboards_-_display_lcd_1_3_240_x_240_pixel_st7789-296742)

FHW-Appel, 2025-04-25:
> ## Beschreibung
> Um den Zustand des Fahrzeugs anzuzeigen (z. B. ob es eingeschaltet ist), benötigen wir eine Möglichkeit zur Ausgabe von Fahrzeugdaten. Zwei mögliche Optionen:
> 
> * **LED-Anzeige**: Eine einfache LED, die den aktuellen Zustand des Fahrzeugs visualisiert (z. B. ein- oder ausgeschaltet).
> * **Kleines Display**: Ein Display, das detailliertere Informationen über den Fahrzeugstatus liefert. Neben dem Fahrzeugzustand könnten hier auch Fehlermeldungen und andere relevante Fahrzeugdaten angezeigt werden.
> 
> Das Display bietet mehr Flexibilität, insbesondere bei der Ausgabe von Fehlermeldungen und der Möglichkeit, weitere Fahrzeugdaten zu visualisieren.
> 
> ## Akzeptanzkriterien
> * **Fahrzeugzustand**: Anzeige des aktuellen Zustands des Fahrzeugs (z. B. "eingeschaltet" / "ausgeschaltet").
> * **Akkuzustand**: Anzeige des aktuellen Akkustands (z. B. prozentualer Wert).
> * **"Run"-Status**: Anzeige, ob sich das Fahrzeug im "Run"-Modus befindet.
> * **Fehlermeldungen**: Möglichkeit zur Anzeige von Fehlermeldungen oder Warnungen, falls diese auftreten.
> * **Benutzerfreundlichkeit**: Die Anzeige sollte leicht verständlich und intuitiv sein, unabhängig davon, ob eine LED oder ein Display verwendet wird.
> * **Platzbedarf**: Die gewählte Lösung sollte möglichst wenig Platz einnehmen, um die Integration ins Fahrzeug zu erleichtern.
> * **Zuverlässigkeit**: Die Anzeige sollte zuverlässig arbeiten, ohne häufige Ausfälle oder Fehler.

Zu jedem Kriterium muss angegeben werden, ob das Kriterium erfüllt wurde, unter Angabe der Evidenz (Nachweis).

closed 2025-04-25 by FHW-Appel