FHW-Appel/MicroMouse-shield #10: Motor Festlegen

FHW-Appel, 2025-04-11:


FHW-Appel, 2025-04-13:
# NEMA08-01
## unverbindlicher Vorschlag
- Schrittmotor
- 3,9 V 0,6 A
- 1,8 Grad pro Schritt
- 3000 Umdrehungen pro Minute
- 20,80 € Reichelt (Beschaffung möglich) 
https://www.reichelt.de/de/de/shop/produkt/schrittmotor_nema_08_1_8_0_6_a_3_9_v-335316?PROVID=2788&gad_source=1&gclid=CjwKCAjwwe2_BhBEEiwAM1I7sZfXDzvBCq_a_o6rtzhl_UpihIC4HqMTLpU1I1Xp7YifUA834RA5IBoCSj8QAvD_BwE

FHW-Appel, 2025-04-14:
# ACT 16HS2404LP1X
## unverbindlicher Vorschlag
- Schrittmotor
- 20 x 39 x 39 mm (sehr schmal)
- 1,8 Grad pro Schritt
- ? Umdrehungen pro Minute
- 8,75 € Reichelt (Beschaffung möglich)
https://www.reichelt.de/de/de/shop/produkt/schrittmotor_nema_16_1_8_8_4_v-260760

FHW-Appel, 2025-04-14:
# NEMA-FLAT02
## unverbindlicher Vorschlag
- Schrittmotor
- 9,5 x 33 x 40 mm (sehr schmal)
- 1,75 V 0,5 A
- 1,8 Grad pro Schritt
- ? Umdrehungen pro Minute
- 22,50 € Reichelt (Beschaffung möglich)
https://www.reichelt.de/de/de/shop/produkt/schrittmotor_nema_flat02_1_8_0_5_a_1_75_v-335315

MathisFeldmann, 2025-04-23:
Für den Motor haben wir uns einen der Vorschläge rausgesucht und den passenden Motortreiber dazu gewählt. Dieser kann zwei Motoren auf einmal steuern. Für ein Fahrzeug benötigen wir also vier Motoren und zwei Motortreiber.

**Verwendete Komponenten:**

### 🔧 Motor – Schrittmotor NEMA FLAT02

Ein kompakter Schrittmotor mit einem Schrittwinkel von 1,8°, einer Stromaufnahme von 0,5 A und einer Nennspannung von 1,75 V. Ideal geeignet für präzise Steuerung in kleinen Robotik-Anwendungen.

[🔗 Zum Produkt bei Reichelt](https://www.reichelt.de/de/de/shop/produkt/schrittmotor_nema_flat02_1_8_0_5_a_1_75_v-335315)

### ⚙️ Motortreiber – TB6612FNG,C,8,EL

Ein effizienter Motortreiber von Toshiba, der bis zu zwei Motoren gleichzeitig steuern kann. Damit reichen zwei dieser Treiber für ein Fahrzeug mit vier Motoren aus.

[🔗 Zum Produkt bei DigiKey](https://www.digikey.de/de/products/detail/toshiba-semiconductor-and-storage/TB6612FNG-C-8-EL/1730070?srsltid=AfmBOoozsnQe4tdp-Cgohsvon74Md1hvjt9q7buxX0vKGaMZHDx1sNPQ)

FHW-Appel, 2025-04-25:
> ## Beschreibung
> Wir benötigen 4 geeignete Motoren für das Fahrzeug. Am besten sollten Stepmotoren in Betracht gezogen werden.
> 
> ## Akzeptanzkriterien
> * Der Motor sollte so kompakt wie möglich sein.
> * Der Motor sollte käuflich erwerbbar sein.
> * Der Motor sollte im Rahmen des verfügbaren Budgets liegen.
> * Der Motor sollte effizient arbeiten.
> * Der Motor sollte die Output-Spannung des Treibers problemlos vertragen.

Zu jedem Kriterium muss angegeben, ob das Kriterium erfüllt wurde, unter Angabe der Evidenz (Nachweis).

FHW-Appel, 2025-04-25:
> 4 Effizienz: Der Motor sollte eine hohe Effizienz bieten.  1,75V und 0,5A

Das ist eine subjektive Betrachtung, Das Kriterium ist nicht zu erfüllen, da sich hohe Effizienz nicht messen lässt.

closed 2025-04-25 by FHW-Appel