FHW-Appel/MicroMouse-shield #7: Ifarotsensor

FHW-Appel, 2025-04-11:


FHW-Appel, 2025-04-13:
#  GP2Y0A51SK0F
## Unverbindlicher Vorschlag
- 2 bis 15 cm
- 5 ms Updaterate
- 6,44 €
- Digikey (Beschaffung umständlich)
https://www.digikey.de/de/products/detail/sharp-socle-technology/GP2Y0A51SK0F/4103863?gclsrc=aw.ds&&utm_adgroup=Sensors%2C%20Transducers&utm_source=google&utm_medium=cpc&utm_campaign=Shopping_Product_All%20Products&utm_term=&productid=4103863&utm_content=Sensors%2C%20Transducers&utm_id=go_cmp-1480315419_adg-60013936729_ad-282983082216_pla-917894487081_dev-c_ext-_prd-4103863_sig-CjwKCAjwwe2_BhBEEiwAM1I7sUA5hvqGb9w29SD-BlHzGAbtqVH6gsy8DFcxUtK0aVst2BgtOzC8choCKEYQAvD_BwE&gad_source=1&gclid=CjwKCAjwwe2_BhBEEiwAM1I7sUA5hvqGb9w29SD-BlHzGAbtqVH6gsy8DFcxUtK0aVst2BgtOzC8choCKEYQAvD_BwE&gclsrc=aw.ds

FHW-Appel, 2025-04-13:
# GP2Y0A41SK0F
## unverbindlicher Vorschlag
- 4 bis 30 cm
- 5 ms Updaterate
- 11,15 €
- RS Components (Beschaffung einfach)
https://de.rs-online.com/web/p/reflexions-lichtschranken/6666568

OmarHas18, 2025-04-23:
--- # Scharfer GP2Y0A51SK0F Abstandssensor ## Übersicht Der **Sharp GP2Y0A51SK0F** ist ein kompakter optischer Abstandssensor, der Entfernungen im Bereich von **20 mm bis 150 mm** messen kann. Er liefert ein analoges Spannungssignal, das proportional zum gemessenen Abstand ist. - **Modellnummer**: GP2Y0A51SK0F - **Größe**: 27,0 mm × 10,8 mm × 12,0 mm - **Messbereich**: 20 mm bis 150 mm - **Versorgungsspannung**: 4,5V – 5,5V - **Stromaufnahme**: ca. 12 mA - **Ausgang**: Analoge Spannung (höhere Spannung bei geringerem Abstand) ## Anwendung Typische Anwendungen umfassen: - Hinderniserkennung in Robotern - Abstandsmessung bei beweglichen Plattformen - Sicherheits- und Überwachungssysteme ## Pinbelegung | Anheften | Beschreibung | |:----|:----------------------| | Nr. 1 | Vcc (4,5 V – 5,5 V) | | Artikel 2 | GND (Masse) | | 3 | Vo (Analoges Ausgangssignal) | ## Schaltung (Beispiel Arduino) '''Klartext Sensor Vcc -> Arduino 5V Sensor GND -> Arduino GND Sensor Vo -> Arduino Analog Pin (z.B. A0) ''' ### Beispielcode (Arduino) ''CPP const int sensorPin = A0; Sensor an A0 angeschlossen void setup() { Seriennummer.begin(9600); } void loop() { int sensorValue = analogRead(sensorPin); Erhaltungsspannung = SensorWert * (5,0 / 1023,0); Serial.print("Sensorwert: "); Serial.print(sensorWert); Serial.print(" - Spannung: "); Serial.print(Spannung); Serial.println("V"); Verspätung(100); } ''' ## Ausgangsspannung vs. Abstand (typisch) Die Ausgabe ist **nicht linear**. Typischerweise gilt: | Abstand (mm) | Ausgangsspannung (V) | |:------------:|:--------------------:| | 20 | ca. 2,8 | | 50 | ca. 1,5 | | 100 | ca. 0,7 | | 150 | ca. 0,4 | > **Hinweis**: Für präzisere Abstandsermittlung empfiehlt sich eine Kalibrierung durch Messung mehrerer Punkte und ggf. eine Umrechnung (z.B. mit einer Polynom-Approximation). ## Weitere Informationen - **Datenblatt**: [Sharp GP2Y0A51SK0F PDF (offizielle Quelle)](https://www.pololu.com/file/0J1970/GP2Y0A51SK0F.pdf) - **Produktlink**: [Digikey Produktseite](https://www.digikey.de/de/products/detail/sharp-socle-technology/GP2Y0A51SK0F/4103863) ---

OmarHas18, 2025-04-23:
# Sharp GP2Y0A51SK0F - Optischer Abstandssensor

## Übersicht
Für das Projekt wird ein Abstandssensor mit folgender Spezifikation verwendet:

- **Modell**: Sharp GP2Y0A51SK0F
- **Messbereich**: 20 mm – 150 mm (2 – 15 cm)
- **Typ**: Optischer, analoger Abstandssensor
- **Größe**: 27.0 mm × 10.8 mm × 12.0 mm
- **Versorgungsspannung**: 4.5V – 5.5V
- **Stromaufnahme**: ca. 12 mA
- **Produktlink**: [Sensor bei Digikey](https://www.digikey.de/de/products/detail/sharp-socle-technology/GP2Y0A51SK0F/4103863)

closed 2025-04-25 by FHW-Appel

OmarHas18, 2025-05-09:
# GP2Y0AF15X
## 
- 0.59" ~ 5.91" (1.5 ~ 15cm)
- 5 ms Updaterate
- 6,81 €
- Digikey
https://www.digikey.de/en/products/detail/sharp-socle-technology/GP2Y0AF15X/9741635?msockid=2131436906e2673e2c65564607696683