FHW-Appel/MicroMouse-shield #8: Gyro

FHW-Appel, 2025-04-11:


FHW-Appel, 2025-04-22:
Welche Achsen müssen überwacht werden?

mheinzebnetza, 2025-04-22:
Mit diesem Sensor soll nur die Rotation in der z-Achse überwacht werden, wenn die bewegung der Mouse in der xy-Ebene erfolgt. Diese Überwachung soll dazu dienen Fehler durch unterschiedlichen Grip bei den Rädern oder ungenauigkeiten bei den Drehungen oder beim Aufstellen auszugleichen

MathisFeldmann, 2025-04-23:
## 🧭 Sensor-Auswahl: MPU6050

Als Sensor haben wir uns für den **MPU6050** entschieden.  
Dabei handelt es sich um einen kombinierten **Beschleunigungs- und Lagesensor (Gyroskop)**. Der Sensor eignet sich besonders gut für Anwendungen, bei denen sowohl die Bewegung als auch die Ausrichtung eines Objekts erfasst werden sollen – zum Beispiel zur Stabilisierung oder Navigation eines Fahrzeugs.

🔗 [Zum Produkt bei DigiKey](https://www.digikey.de/de/products/detail/tdk-invensense/MPU-6500/4385413?gclsrc=aw.ds&&utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=PMax%20Shopping_Product_High%20ROAS&utm_term=&productid=4385413&utm_content=&utm_id=go_cmp-20188050700_adg-_ad-__dev-c_ext-_prd-4385413_sig-CjwKCAjwn6LABhBSEiwAsNJrjkaaMHju4PiSNp-ekII3FRmDZyAnH4WMiYG736heD2F1TDkez8iMehoCWN4QAvD_BwE&gad_source=1&gclid=CjwKCAjwn6LABhBSEiwAsNJrjkaaMHju4PiSNp-ekII3FRmDZyAnH4WMiYG736heD2F1TDkez8iMehoCWN4QAvD_BwE)

FHW-Appel, 2025-04-25:
Bitte ändern Sie den Namen des Issues. Es sollte aus dem Namen des Issues klar hervorgehen, was die Aufgabe ist.

closed 2025-04-25 by FHW-Appel