FHW-Appel/MicroMouse-shield #9: Tabelle mit Pinout erstellen

FHW-Appel, 2025-04-11:
Im Wiki erstellen

FHW-Appel, 2025-05-14:
Wenn noch Pins frei sind, dann bitte für weitere Abstandssensoren einteilen.

FHW-Appel, 2025-05-14:
Wenn noch möglich, dann UART Schnittstelle routen.

FHW-Appel, 2025-05-16:
Warum hat die Liste "_Pinout Mikrocontroller Nucleo32_" mehr Einträge als die Liste "_Nötige Pins für den µC_"?

Warum wird die eine Liste mit 0 beginnend nummeriert?

Können beide Listen nicht miteinander verschmolzen werden?

In der Liste "_Pinout Mikrocontroller Nucleo32_" sollte im jeden Fall auch die Spalte "_zu_" hinzugefügt werden.

Alexanderliebert, 2025-05-20:
Absprache mit SW-Team
V_in ist ein PIN, aber 

Es ist zur Zeit nur ein Motortreiber in der Liste "notwendiger Pins Liste" vorhanden. Weitere Pins zu Ansteuerung benötigt/ Absprache benötigt, wie diese Angesprochen werden sollen, da man nur eine begrenzte Anzahl von Pins hat.

Eine genaue Absprache, wie die Pins miteinander verbunden sind z.b alle STBY zusammen verbunden. 

D7 Pin/ PC15(14) genauer angucken ob dieser mit hohem Strom funktioniert.

Verwechslungsgefahr der Mikrocontroller jetziger Mikrocontroller ist der [Nucleo-L011K4](https://www.st.com/en/evaluation-tools/nucleo-l011k4.html#overview) und [STM32L011K4
](https://www.st.com/en/microcontrollers-microprocessors/stm32l011k4.html#overview) ist der Prozessor

closed 2025-06-27 by FHW-Appel