FHW-Appel/MicroMouse-shield #12: Stückliste erstellen und Händler heraussuchen

mheinzebnetza, 2025-04-25:
# Beschreibung

Um die Micromouse zu bauen ist es notwendig eine Stückliste zu erstellen, damit die nicht vorhandenen Bauteile eingekauft werden können. 
____

# Akzeptanzkriterien
* __Verfügbarkeit__: Die Bauteil muss bei einem Händler auf Rechnung bestellt werden können.
* __Vollständigkeit__: Die Liste muss alle Teile enthalten, die für den Bau und die Bestückung der Micromouse notwendig sind.
* __Listenform__: Es sollte eine Liste erstellt werden, die per mail verschickt werden kann.
____
# Liste

FHW-Appel, 2025-05-16:
Für die Stückliste bitte ebenfalls eine Wikiseite erstellen. Folgende Spalten sollte die Stückliste enthalten:
Bezeichner, Value, Hersteller, Herstellernummer, Distributor, Bestellnummer, Anzahl, Kosten je Stück.

mheinzebnetza, 2025-05-23:
Wikiseite wurde erstellt, die Befülllung der Liste folgt peu à peu.